lasselukkari/DuinoDCX #25: Static files not found

samovesel, 2018-09-28:
Hi,
sorry to bother you, but I can't get to compile sketch right. I followed your instructions, and everything seems OK, but when I upload it to ESP32 looks like there are no files in /static folder, so the only thing I can do is login and then it stops, since I can't load  /static/css/main.c8aa8249.css and /static/js/main.aa0c02d7.js.
manifest.json & favicon.ico are present.
The operating system is Win 7. 
Samo

samovesel, 2018-09-28:
OK, I just needed to replace backslashes with slashes in void ServeStatic in StaticFiles.h and then everything is OK. Is there something I 'm doing wrong during building?

lasselukkari, 2018-09-28:
I actually most likely know the problem already. It's because paths in windows use '\' rather than '/'. I will update the tooling to fix this. Should get it done tonight.

lasselukkari, 2018-09-28:
I updated the library that generates the static payloads. Can try again with the latest version in the master branch?

samovesel, 2018-09-28:
First backslash is now ok, but the problem is still there:

void ServeStatic(WebApp* app) {
  app->get("asset-manifest.json", &static_asset_manifest_json);
  app->get("favicon.ico", &static_favicon_ico);
  app->get("", &static_index);
  app->get("manifest.json", &static_manifest_json);
  app->get("static/css\main.c8aa8249.css", &static_static_css_main_c8aa8249_css);
  app->get("static/js\main.99370a8c.js", &static_static_js_main_99370a8c_js);
  app->get("static/media\glyphicons-halflings-regular.448c34a5.woff2", &static_static_media_glyphicons_halflings_regular_448c34a5_woff2);
  app->get("static/media\glyphicons-halflings-regular.89889688.svg", &static_static_media_glyphicons_halflings_regular_89889688_svg);
  app->get("static/media\glyphicons-halflings-regular.e18bbf61.ttf", &static_static_media_glyphicons_halflings_regular_e18bbf61_ttf);
  app->get("static/media\glyphicons-halflings-regular.f4769f9b.eot", &static_static_media_glyphicons_halflings_regular_f4769f9b_eot);
  app->get("static/media\glyphicons-halflings-regular.fa277232.woff", &static_static_media_glyphicons_halflings_regular_fa277232_woff);

lasselukkari, 2018-09-29:
True. I should have known this... I have now changed it to `relativePath.replace(/\\/g, '/')`. I really should write unit tests for the tooling or alternatively stop coding drunk on friday nights and go out. Can you try again.

samovesel, 2018-09-29:
Everything is perfect now. And that mute / unmute toggle button is really useful. Thank you!

closed 2018-09-29 by samovesel