lasselukkari/DuinoDCX #65: Bessel filter plot

samovesel, 2019-05-27:
Hi,
it seems that there is something wrong with crossover plot formula for bessel filter. Center frequency is just off for bes12 & bes24. Or am I missing something?

Best!

[image]
[image]

lasselukkari, 2019-05-28:
Thanks for reporting. I will investigate this.

closed 2019-05-30 by lasselukkari

lasselukkari, 2019-05-30:
I consulted Ilkka Huhtakallio, the mastermind behind the the transfer function code. He was quickly able to pinpoint the problem.

samovesel, 2019-05-30:
Perfect, thank you! 
I know it might be overengeenering, but I have one more suggestion for the plot.
It would be really nice to have combined freq response curve (maybe also with included eq curves) on crossover plot. 
Not to mention phase response.
That would be really useful, maybe even as standalone project, since it's really annoying to find that kind of stuff online.
But, like I said, it's probably not in the scope of duinoDCX.

Best!