lasselukkari/DuinoDCX #69: Not able to reach 192.168.4.1

hackschnitzel09, 2019-06-14:
Hi,
i have installed v0.0.25 update progress runs untill 100% also the ESP32 generates a hotspot called DCX2496 and am able to log in. But on mobile devices the wifi isnt stable and kicks me out after a few seconds. On my PC it shows as connectes but is not able to reach 192.168.4.1 .

I did the instalation like in the step-by-step instructions. Also I changed the Partition Scheme to Minimal SPIFFS.

What can I do to get the thing running?

Thank you for the nice projekt.
Regards

juanjosemmoraless, 2019-06-14:
I have the same problem,hotspot wifi unstable, i checked sígnal with inssider software.

lasselukkari, 2019-06-16:
Hello

As you probably can guess I don't see these problems myself and there are many others who have the project working just fine. It would really help if you would describe your setup in more detail. For example providing your mobile phone brand and os version and so on. Also it would help to know the controller physically located.

Unfortunately the ESP32 Arduino core still has a lot of bugs. I have had to downgrade the version back to old one because of new bigger bugs in the later versions, but the whole esp32 Arduino project seems to be still quite unstable. For example there are some know issues with specific phones and operating systems. I'm hoping they would fix these problems in a update but so far they have just made the problem worse.

You can try also to: 
- power the controller on and off
- test other phones or computers
- test another usb charger
- use an RS232 extension cable so that the DuinoDCX has better wifi range
- connect to the duinodcx when it is not plugged to the Ultradrive
- install older versions of the DuinoDCX
- disable mobile data connection on your phone
- try it in another location

lasselukkari, 2019-06-16:
One person had similar problems at some point. Although it does not make much sense he reported that he got it fixed when he used the ESP32 Arduino core version 1.0.0 to compile the initial version of the OTAWebUpdater firmware and then uploaded the binary version of the DuinoDCX. The ESP32 arduino core version can be changed in the _Boards manager..._ menu of the Arduino IDE:

hackschnitzel09, 2019-06-23:
Hi, 
i downgraded the ESP32 Arduino core vision to 1.0.0 and it worked. Thanks for this advice. And a very big thanks for the nice and easy solution to get the DCX controllable by mobilephones. 
THANKS!:)
Kind regards

lasselukkari, 2019-06-23:
Thanks for reporting the problem. The correct core version is now mentioned in the documentation.

closed 2019-06-23 by lasselukkari