lasselukkari/DuinoDCX #85: DCX Wifi problem

Jost9696, 2019-08-06:
Hello, 
I want to install everything into the DCX with a switch. I have already a small 5V Powersupply and this chip https://www.ebay.de/itm/MAX3232-ESE-RS232-auf-TTL-Converter-Module-Konverter-Board-6-St%C3%BCck-piece/173358084088?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

I conntecte the TX and RX from the ESP32 to the small chip and the 3.3 V. the cabel to the Controller only is connected with GND RX and TX. In my fault I try to short RTS and CTS and now I killed my EEprom i think. I ordered a new one already. The software works so i can see searching on my phone. But there is no connection.

lasselukkari, 2019-08-06:
Shorting the RTS and CTS should not cause problems. Infact linking input and output of the RS232 chip can be used to verify the electrical connections. When the output goes high the input should read high as well.

I will post you an Arduino sketch later today that you can use to verify that your electrical connection are working as intended. You will need a multimeter to complete the test.

lasselukkari, 2019-08-06:
Also verify that you have the correct setting selected in the Ultradrive for the `SETUP -> MISCELLANEOUS -> PORT` setting.

[image]

lasselukkari, 2019-08-06:
I have added a [HarwareTest sketch](https://github.com/lasselukkari/DuinoDCX/blob/master/HardwareTest/HardwareTest.ino) to the project.

Jost9696, 2019-08-19:
Now the test works everything shown as okay in the serial monitor. Thank you. (Y)

closed 2019-08-19 by Jost9696

Corylus-Core, 2020-01-04:
My Setup is also stuck in "Searching". Already tried to change RX and TX but nothing happened. I saw that there is a HardwareTest.ino which i've tried with looping the RX and TX pins on the cable to the DCX2496, but i don´t understand how i get the readout from you HardwareTest.ino. Sorry for my stupid questions :-( . Sincerely Georg

lasselukkari, 2020-01-05:
Hi.

Could you take a picture of your setup and a give a full list of components in your wiring chain? I may be able to help.

To use the HardwareTest.ino open it in the Arduino IDE to flash it to the board. Then open `Tools -> Serial Monitor` and set the baud rate to `115200`. You will now see the text `Error` or `OK` in output repeated every 2 seconds.