lasselukkari/DuinoDCX #135: Please Help

Hakuna-23, 2019-10-13:
Hello first of all thank you for this great shit :)
i hope you can help me with my problem .
I do everything like in the guide and all works well during the point i want to start the remote control after entering the username and password it is only searching the howl time and i can not to anything. Do you have an idea what is maby wrong?

thank you very much

[image]

killerops, 2019-10-13:
I also have this issue on iOS, but on chrome in Android works well.
Tested on Chrome on iPhone7 and iPad2.

Hakuna-23, 2019-10-14:
OK I tried it on chrome on Android but the same problem :(

lasselukkari, 2019-10-14:
Hi. There is an indication of two possible problems in your screenshot. 

1. The `Searching...` text shows up when the the DuinoDCX controller is not able to communicate with the Ultradrive using the RSR232 port. This is most likely caused by bad wiring or problems with the Ultradrive `PORT` settings.
2. The `Check network connection` message pops up when the browser is not able to make requests to controller. Is this message constantly there or did it popup after you disconnected from the wifi network create by the DuinoDCX?

I will try to prepare a better step by step troubleshooting guide this week. In the meantime it would be really nice if you could update the bug report to use this [template](https://raw.githubusercontent.com/lasselukkari/DuinoDCX/master/.github/ISSUE_TEMPLATE/bug_report.md).

Previously the Arduino ESP32 core has had problems with some operating systems, mostly with Windows. These should have been now finally fixed since the core version 1.0.3.

lasselukkari, 2019-10-14:
I just realized the hardware test sketch may have given false positives in some situations. Could you both verify again with the updated [HardwareTest](https://github.com/lasselukkari/DuinoDCX/blob/master/HardwareTest/HardwareTest.ino) that your hardware configuration is working as expected. The instructions are included in the comments.

lasselukkari, 2019-10-14:
I also today tested the app on different configurations and found a bug in the webserver header handling. This is now fixed in the latest release.

I tested the setup successfully with latest versions of:
* Mac + Chrome
* Mac + Firefox
* Mac + Safari
* Android + Chrome
* Ubuntu + Chrome
* Ubuntu + Firefox
* Windows 7 + Chrome
* Windows 7 + Edge

lasselukkari, 2019-11-14:
I'm closing this due to inactivity for one month. Feel free to reopen the issue if needed.

closed 2019-11-14 by lasselukkari