lasselukkari/DuinoDCX #166: Saving profiles directly over the DuinoDCX Frontend

Corylus-Core, 2019-12-13:
If we could save profiles directly over the DuinoDCX frontend this would be great. Thanks for your great effort to our old an beloved DCX2496. With your Project it is like having a completely new device. Thanks!

lasselukkari, 2019-12-16:
This feature request seems to pop up quite often. I will take a look what it would take to get this implemented.

lasselukkari, 2020-01-11:
Unfortunately with my current understanding of the settings data binary format and the transfer protocol applied on top of it  this does not seem to be possible. I wish some one from Behringer would chime in here.

If someone wants to help take a look at the binary files the original remote software creates when you save the settings to a local file. To get the current state I have created a manual lookup system that just reads the values from predefined locations in similar binary data that defines the current settings. If you think you can help I can provide more details.

Lyteral, 2020-02-09:
This would be also insanely interesting for me as you could just save presets on your device and upload them via DuinoDCX as you need them.

One way to go at this (I believe) would also be to simply save all the user-editable settings in a JSON / XML file when saving the profile and when uploading the profile, you just iteratively send all settings to the DCX, as if someone would set them up via the GUI..
This would not require understanding of the proprietary Behringer binary format.

I don't know however how easy and/or reliable this implementation would be

lasselukkari, 2020-03-22:
> One way to go at this (I believe) would also be to simply save all the user-editable settings in a JSON / XML file when saving the profile and when uploading the profile, you just iteratively send [...]

If I remember correctly there are almost 1000 settings in the DCX. I believe setting all of them one by one would be too slow.

But this would be maybe doable for output settings. The output settings are resetted by the DCX if the routing is changed for example from mono -> 3 way stereo. This way only the settings that are not default would be required to be updated if the routing mode is set first.

Lyteral, 2020-03-23:
> But this would be maybe doable for output settings. The output settings are resetted by the DCX if the routing is changed for example from mono -> 3 way stereo. This way only the settings that are not default would be required to be updated if the routing mode is set first.

Good idea! So you would essentially have (or set) a "base setting" (which is one of the individual routing profiles) and you would then only iteratively set all output settings one by one?
I think for saving and uploading quick "profiles" i.e. for different sound system set-ups that would be very well suited.

c-guenther, 2023-10-25:
I would like to second that. Storing/Recalling profiles would be great as well as the idea stated above.

HumbleDeer, 2025-06-06:
If I end up putting my DCX to use or in service again, I might actually end up investigating this. I suspect the binary file is formed by  more or less directly translating Clang-style struct(s) with concatenated struct entries of either a fixed length or a variable length defined by a header of the struct entry. It's a common way of structuring proprietary file formats.