lasselukkari/DuinoDCX #167: A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

Corylus-Core, 2019-12-17:
I´m unable to upload the Configuration to my ESP32 board. I´m always getting the error message "A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header"

lasselukkari, 2019-12-17:
For some reason your development board is not responding to the command that should initialize the upload process. There is either a problem with the serial port connection or the board fails to boot in the flash mode.

Some development boards have a faulty reset circuit for selecting the boot mode automatically. If this is the problem you can perform the boot mode selection manually:
1. Hold down the `BOOT` button
2. Press and release the `EN` button
3. Release the `BOOT` button
4. Upload the project

Try googling with the exact phrase and you will get plenty of [results](https://www.google.com/search?q=%22Failed%20to%20connect%20to%20ESP32:%20Timed%20out%20waiting%20for%20packet%20header%22).

See also the discussion in the ESP32 Arduino core project [issues](https://github.com/espressif/arduino-esp32/search?q=Failed+to+connect+to+ESP32%3A+Timed+out+waiting+for+packet+header&type=Issues). 

This is not a problem with this particular project. Unless you happen to have the [SerialChiller](https://github.com/lasselukkari/SerialChiller) board I can not provide general hardware debugging help as my time is limited.

closed 2019-12-17 by lasselukkari

Corylus-Core, 2019-12-17:
It was much easier then i thought. Some people had to solder a bridge on their boards, i only had to hold the boot button when i pressed upload.

lasselukkari, 2019-12-17:
Nice to hear you got it working!