lasselukkari/DuinoDCX #187: Connecting to wifi network issue

AM555, 2020-02-10:
Hi Lasse,

I have reposted this here as requested.

I made a batch of your DCX remotes and I have managed to get everything working except connecting via an existing wifi network. Connecting directly to the dongle works fine but I can't manage it via my home router or a hotspot on my phone. Home router just won't connect and it throws up an error message. Via my phone hotspot it seems to connect fine and my phone tells me it is connected but chrome won't connect to load the GUI. I have also tried to change the authentication type of my home router with no success.

Let me know if you need any more info.

Thank you for your time.

Kind regards,
Matt

lasselukkari, 2020-02-10:
Please try to upload this sketch: https://raw.githubusercontent.com/espressif/arduino-esp32/master/libraries/WiFi/examples/SimpleWiFiServer/SimpleWiFiServer.ino Change the wifi credentials to mach your wifi router. Does this work?

Have you tried to connect it to another wifi router other than your phone? What is the brand and model of your wifi router.

When connected to the phone hotspot have you disabled your mobile internet connection? What url address are you using when in hotspot mode? With android you need to use your local ip.

AM555, 2020-02-15:
I have uploaded that sketch and have managed to connect to my home router and phone hotspot successfully. I am unsure what I do next, which page do I need to load?

I have only tried my phone and my home wifi router. My home wifi router is branded with the make: EE and model: bright box (r) wireless router so I am unsure of the actual make and model. It would probably be prudent to try to find another router to try it with.

I have set my home router to WPA but my phone hotspot will only allow me to use WPA2 PSK. 

I have disabled my internet data when connected to my hotspot. When loading the drinoDCX sketch I wasn't using the phone directly I was trying to use my laptop also connected to the hotspot and the ip 192.168.4.1.

lasselukkari, 2020-02-15:
When you upload that sketch and watch the serial monitor window it will print the local ip address when it gets a connection. Copy this ip to your web browser address bar and hit enter. You should see a simple website with two links. The links will not do anything but if the page loads this will show if the problem in the esp32 Arduino core or is the problem in this code base. If you get the page loading the problem is in this project rather in things outside of my control.

That IP address you are using is the IP that only works when connected directly to the DuinoDCX in access point mode. When you connect the DuinoDCX to another access point like the phone hotspot it will be assigned an IP address from that network. The IP address is visible the user interface:

[image]

Use this IP when you access the device using the phone itself or from the computer.

AM555, 2020-02-27:
Sorry I haven't replied quicker.

For some reason the serial monitor wounldn't print when I uploaded the SimpleWiFiServer.ino   sketch so I couldn't find out the IP. I tried this on two machines with the same result.

I did test the serial monitor with this sketch:

static const int heartbeat = 13;

void setup() {
  Serial.begin(9600);
  pinMode( heartbeat, OUTPUT );
  }

void loop() {
  digitalWrite( heartbeat, HIGH);
  Serial.println("Hello");
  digitalWrite( heartbeat, LOW);
  delay(1000);
}

It did work and printed "hello" repeatedly in the serial monitor once I had unplugged and re-plugged the project.

I have come to a conclusion though. Once I uploaded the SimpleWiFiServer.ino and logged into my home router I could see the IP of the project attached. The web page did load with the following:

Click here to turn the LED on pin 5 on.
Click here to turn the LED on pin 5 off.

lasselukkari, 2020-02-27:
Sounds like you did not set the correct serial port baud rate in the serial monitor. Your example uses 9600 and the SimpleWiFiServer.ino uses 115200. If this is wrong you either see nothing or garbage.

So what happens when you try to connect the DuinoDCX with the ui to the router using those same wifi credentials? Do you get an error? If you do, where and what does it say? Just to make sure nothing else is wrong with your setup remove the RS232 adapter and use the ESP32 development board without anything connected to it.

AM555, 2020-02-28:
Sooooooooo,

I have had success using my phone hotspot, what I didn't initially realise was I had to use a different IP address like you described with the screen shot, once I do that it works as it should.

I still can't connect via my home Wifi router. When I try to I get the "could not connect to network" red error message hat pops out in the bottom left hand corner of the GUI (this is the error message I was talking about before). It is becoming more probable that it is a compatibility issue with my home router. I will have to try to get my hands on another one and see if I can replicate the problem.

AM555, 2020-02-28:
P.S. it wasn't a baud rate issue, I changed it to 9600 and back between the two sketch's.

TomD8503, 2020-03-13:
I had some issues with connecting to wifi. It sometimes worked, sometimes not. After some troubleshooting it turned out to be a bad power adapter. After switching to a Apple 12W 2,4A USB power adapter (iPad charger) all the problem are gone.

lasselukkari, 2020-04-04:
Closing due to inactivity for over a month.

closed 2020-04-04 by lasselukkari