lasselukkari/DuinoDCX #237: searching

davidd1205, 2020-06-16:
159/5000
Hello
[image]
[image]
[image]
[image]
[image]

I'm stuck on the searching page
Last update
dcx firmware latest 1.18 model 2016
esp32s
tested with hardware but ok even without bridge

lasselukkari, 2020-06-16:
The Ultradrive model you have should work just fine. Did you verify your setup with a multimeter? If not, do that first before we go any deeper in with the debugging.

davidd1205, 2020-06-16:
thanks a lot 

i have tested on esp32s and the voltage rx tx is ok

but on a max232 i don't have a voltage 
i think my max232 is fault 

i dont have another for test

closed 2020-06-27 by lasselukkari