lasselukkari/DuinoDCX #244: i get just "Searching..."

MiroslavSPK, 2020-07-21:
hi,
i have a problem, it do not connect to my DCX2496.
i get only "searching" all the time.
i have the older behringer DCX2496 with the pcmcia card reader.
use firmware 1.17, but tried even 1.16
the hw test passed "OK"
pin 2 voltage fluctuate between -5V and 5V (exactly around +/- [5.5-6V)
i uploaded even diferend versions of DuinoDCX to eliminate some version problems... (v0.0.37, v0.0.36, v0.0.31, v0.0.29)
is the problem mi old dcx2496?

conecting ipads, iphone, pc all the same result.

when i connect it to my laptop with a cheap chinese usb-serial cable it comunicates ok.

thanks for your help
[image]

lasselukkari, 2020-07-21:
Hello!

There should be no reason the model you have would not be compatible. So far all models have worked around the world. I have the old version with the pcmcia reader too. Also your wiring and other components seem to be correct.

Did you perform the hardware check with your RS232 cable connected? Are you sure you have null modem cable and not  straight through?

MiroslavSPK, 2020-07-21:
ok.
now i feel real stupid... :-D :-D :-D
i was thinking the cable ist just a extension cable, i dont have one...
so there is the problem... i will get one, in few days :-)

one more thing. i see no donation button. where can i donate for your work?

thanks for everithing :-)

MiroslavSPK, 2020-07-23:
now it works :-)
[image]

closed 2020-07-23 by MiroslavSPK