lasselukkari/DuinoDCX #245: No DuinoDcx page access after flashing

Sonik-audiolab, 2020-11-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots or video to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Hardware (please complete the following information):**
 - DuinoDCX firmware version: [e.g. 0.0.25]
 - Development board: [e.g. ESP32-DevKitC V4]
 - RS232 adapter: [link to specs]
 - Photo of wiring

**Development environment (please complete the following information):**
 - Arduino IDE version: [e.g. 1.8.9]
 - ESP32 Arduino core version: [e.g. 1.0.1]
 - Upload speed: [e.g. 921600]
 - Partition Scheme: [eg. Minimal SPIFFS (Large apps with OTA)]
 
**Additional context**
Add any other context about the problem here.

Sonik-audiolab, 2020-11-22:
I forgot...
I follwed step by step .
Everything well except this .

[image]

Sonik-audiolab, 2020-11-22:
...sorry again.
I'm new on this and trying to learn step by step.
Understood that I misse one step.
I change directory to Duinodcx folder where package.json is located.
My Duino folder is like this.
[image]

Now I'm in Duinodcx folder and I launch the two commands
[image]
[image]

Now the Duinodcx folder is like this
[image]

Then I open Duinodcx.ino with Arduino IDE.
I upload sketch.
After this try to connect again to DCX2496 wifi.
When  enter to Ip I get the same stuck page.
[image]

lasselukkari, 2020-11-22:
Please read the [instructions](https://lasselukkari.github.io/DuinoDCX/) again carefully and follow them step by step. Also take a look at the closed issues from the past.  Your comments reveal that you have not done neither of th yet. All the information is there and I would be just repeating myself.

Sonik-audiolab, 2020-11-22:
ok. Thank you for you time spent to answer me.
So I got wrong.
Nodejs process is needed just to modify the code.
Sorry. I've seen to late other issues post closed.
Now I've seen. So I will repeat the sketch upload without making troubles with Nodejs.
In case I will get again problem now I know how to use the testing fil.
For sure I forgot to say you..
Nice job with this kit and software.
Thank you.

Sonik-audiolab, 2020-11-22:
Sorry man. 
So please sorry... I need to ask you 2 things more to confirm that I understood well.
I'm stupid maybe.
1) I didn't understand about test.
Hardware test needs to be done with null cable connected right?
2)It stucks on "Searching" because does not receive signal form DCX2496. Right?

I ordered to build 2 kits. Assembled the other and I get the same result.
Doing hardware test also the same. I get "error".
Tried to upload Duino sketch to booth of them.
I can connect to wifi but the couple stuck in searching.
So, if I understood well, I get error from test and also user page stucks in searching because I bought wrong cables.
They are simple Rs232 serial lineDB9.

lasselukkari, 2020-11-23:
The hardware tests need to be done with the complete setup including the cabling your are planning to plug to the dcx2496. There are two types of cables available: null modem and straight. Do a quick google search and you will see the difference. You need a null modem cable. But because male to male null modem cables are really rare you need to use a male to male gender changer and then female to male null modem cable. You can use a multimeter to detect what type of cable you have bought.

Do you have the gender changer in your setup? If not you are most likely using the wrong type of cable.

Sonik-audiolab, 2020-11-23:
Ok. Thanks a lot. I took a mistake when bought all the parts. I really missed NULL and ordered a serial Rs232 “ male to female”. Today I bought Null cable and it works great in booth kits.
Now ordered null “male to male” adapter to don’t waste the wrong cables.
Again thanks. 
[image]

closed 2020-11-23 by Sonik-audiolab

lasselukkari, 2020-11-23:
I'm glad you got it working.

Sonik-audiolab, 2020-11-23:
Thank you man. Dcx users all around the world have to say you a really big “THANKS!”
It’s really an useful tool to have. Great job.