lasselukkari/DuinoDCX #246: ESP 32 + Wiznet W5500

g-philipp, 2020-11-23:
Ethernet with the Wiznet W5500 Ethernet Board

set up a static ip address and select ethernet instead of wifi in the webinterface 

[image]
[image]

lasselukkari, 2020-11-23:
The webserver library supports any class derived from the Stream so the Wiznet W5500 should work. I personally have not ever used the ESP32 + Wiznet W5500 combination but the library works fine with other arduino boards and ethernet shields.

See the ethernet example here: https://github.com/lasselukkari/aWOT/blob/master/examples/Ethernet/Ethernet.ino

g-philipp, 2020-11-23:
Is it much effort to programm the ethernet function to exist in the settings and work?

lasselukkari, 2020-11-23:
Compared to the WiFi connection management with credential the Ethernet is actually simpler.

g-philipp, 2020-12-22:
well ive tried a couple of things the last days and ive got it kinda working, my problems are that first it loses connection all the time: 
[image] and second, when I try to connect via html I only see the follwing: 
[image]

I of course had to modify the code and I think that the issue is cause by deleting some stuff... The code now looks like this: 
`#include <ESPmDNS.h>
#include <Update.h>
#include <Preferences.h>
#include <SPI.h>
#include <Ethernet2.h>
#include "local_config.h"
#include "aWOT.h"
#include "StaticFiles.h"
#include "Ultradrive.h"

#define VERSION "v0.0.39"
#define BUILD_DATE __DATE__ " " __TIME__

#define DEFAULT_AUTH "Basic RENYMjQ5NjpVbHRyYWRyaXZl" // DCX2496:Ultradrive in base64
#define DEFAULT_SOFT_AP_SSID "DCX2496"
#define DEFAULT_SOFT_AP_PASSWORD "Ultradrive"
#define DEFAULT_MDNS_NAME "ultradrive"
#define DEFAULT_FLOW_CONTROL false
#define DEFAULT_AUTO_DISABLE_AP false

#define RTS_PIN 21
#define CTS_PIN 22

#define RESET_PIN 13

#define WIFI_HOST_NAME "ultradrive"
#define AUTH_KEY "auth"
#define SOFT_AP_PASSWORD_KEY "apPassword"
#define SOFT_AP_SSID_KEY "apSsid"
#define MDNS_HOST_KEY "mdnsHost"
#define FLOW_CONTROL_KEY "flowControl"
#define AUTO_DISABLE_AP_KEY "autoDisableAP"

#define AUTH_BUFFER_LENGHT 200
#define MAX_DEVICES 16
#define POST_PARAM_SSID_KEY "ssid"
#define POST_PARAM_PASSWORD_KEY "password"
#define SSID_MAX_LENGTH 33
#define PASSWORD_MAX_LENGHT 65
#define CONNECTION_TIMEOUT 10000
#define RECONNECT_INTERVAL 20000
#define BASIC_AUTH_LENGTH 200
#define SOFT_AP_SSID_LENGTH 65
#define SOFT_AP_PASSWORD_LENGTH 65
#define MDDNS_NAME_LENGTH 65

const int NTP_PACKET_SIZE = 48;   // NTP time stamp is in the first 48 bytes of the message.
byte packetBuffer[NTP_PACKET_SIZE]; // Buffer for both incoming and outgoing packets.

Preferences preferences;
EthernetServer server(80);
HardwareSerial UltradriveSerial(2);
Ultradrive deviceManager(&UltradriveSerial, RTS_PIN, CTS_PIN);
Application app;
Router apiRouter("/api");

char basicAuth[BASIC_AUTH_LENGTH];
char softApSsid[SOFT_AP_SSID_LENGTH];
char softApPassword[SOFT_AP_PASSWORD_LENGTH];
char mdnsName[MDDNS_NAME_LENGTH];
bool flowControl;
bool autoDisableAP;

char authBuffer[AUTH_BUFFER_LENGHT];
char ssidBuffer[SSID_MAX_LENGTH];
[... 384 lines cut ...]

/*
 * Choose the NTP server pool for your geographical region for best
 * performance (fewer lost packets).
 *
 * *** Uncomment only one of the following "timeServer[]" defines. ***
 */
//const char timeServer[] = "pool.ntp.org";   // Default NTP server pool.
// const char timeServer[] = "africa.pool.ntp.org";   // Regional server pool.
// const char timeServer[] = "asia.pool.ntp.org";   // Regional server pool.
 const char timeServer[] = "europe.pool.ntp.org";   // Regional server pool.
// const char timeServer[] = "north-america.pool.ntp.org";  // Regional server pool.
// const char timeServer[] = "oceania.pool.ntp.org";    // Regional server pool.
// const char timeServer[] = "south-america.pool.ntp.org";  // Regional server pool.
// const char timeServer[] = "time.nist.gov";     // Original example target server (least preferred).

const uint8_t SLEEP_SECS = 15;      // Number of seconds to sleep between queries to the time
            // server. Please don't set this any lower than 10 unless
            // timeServer[] is a local NTP server on -your- network.`

Its my first time using an ethernet board and i dont fully understand your code so forgive me...

Hope you can help me 

~Philipp

lasselukkari, 2020-12-22:
Hi. I have never tested ESP 32 + Wiznet W5500 combination myself. With the ESP32 WiFi the WifiClient destructor stops the client automatically when it goes away from the scope. The EthernetClient in Ethernet2 library does not define a destructor. 

So you will need to at least change the 

```
void processWebServer() {
  EthernetClient client = server.available();
  app.process(&client);
}
```

to

```
void processWebServer() {
  EthernetClient client = server.available();
  if (client.connected()) {
    app.process(&client);
    client.stop();  // add the client stop here
  }
}
```

lasselukkari, 2020-12-22:
I have the same ethernet module somewhere. I could try it out tomorrow if I find it.

g-philipp, 2020-12-24:
with the change in "processWebServer()" its working! But it would be awesome if you could check the code, you may implement this into your code? Merry Christmas!

lasselukkari, 2020-12-25:
I'm glad you got the Ethernet connection working! It sounds like your RS232 connection is not reliable for some reason. This is not normal and I have not seen the connection to behave like that. 

I would try another set cabling first. If that does not work I would try another MAX3232 module. I have read from old forum discussions that the chinese fake copies can be unreliable especially at higher baud rates. You can also make sure it's not the changes that you have made by switching back to the wifi code.

Feel free to create a pull request. I can add the support for the Ethernet to codebase but I need to find my Ethernet module first. I ordered a new one but it will take a couple of weeks to arrive.

lasselukkari, 2021-01-21:
I got a new ethernet module today. Let's see how this goes.

lasselukkari, 2021-01-21:
Is there a reason you are using the Ethernet2 library? The standard one seems to work as well.

g-philipp, 2021-01-21:
The UDP example worked for me but the webserver didnt, also with many other libs i had the same problem, but with the Ethernet2 (there are different Eth2 libs, only one worked for me) ist worked as it should

lasselukkari, 2021-01-21:
And did you get the RS232 working reliably? I bought a module and board that does not expose any extra pins. If you got the rs232 connection working I would not have to test the changes together with the rs232 module. I can hook up the module to another board but it would be more convenient if I do not have to.

g-philipp, 2021-01-21:
No sadly i didnt get it working reliably but I found out, that **sometime** (only sometimes) when the music stops, it "catches itself" and I can use the webinterface like normal, but then if the music starts, it is endless looping with loading.

It is very strange :(

lasselukkari, 2021-01-21:
OK. I will let you know if I get it working properly.

The ESP32 core has a non standard definition for the server that is not compatible with the rest of the Arduino ecosystem. If you want to use the normal Ethernet library you have to do a simple modification to the ESP32 core files.

You need to change this:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Server.h#L28

from

```
virtual void begin(int port) =0;
```

to 

```
virtual void begin() =0;
```

Unfortunately you can not compile the WiFiServer lib anymore with this change until you change it back. The result is that the WiFiServer and EthernetServer can not be used simultaneously. Until the esp32 core gets their shit together I will add this functionality to be configured with a feature flag and there will not be a user interface for setting the static ip.

lasselukkari, 2021-01-22:
BTW have you tried or considered the lan8710/lan8720 for the Ethernet. It seems to be supported out of the box with the Arduino framework. I'm new to Ethernet options with ESP32 and I'm trying to figure out the pros and cons between these two solutions. The Wiznet chips uses SPI and this may be at least in theory blocking the other IO pins under heavy use. This could explain the problems you are seeing with RS232 port stability. I ordered this board for evaluation https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/. Not just because of this project but rather to verify the compatibility with the web server library I have been developing.

g-philipp, 2021-06-17:
Hey, its been a long time since my last post. In the meantime i used the ethernet board and i think the only problem is still that the serial converter sometimes looses connection. But i will not change it. Have you tried some testing with your ethernet board?

IcebergG1, 2024-08-14:
Good day. Did someone find a working solution to connect it over Ethernet?
I would really appreciate a guide for this.
Thank you kindly.

Nuet0815, 2025-11-09:
Hi, did you got the chance to test ethernet with lan8710/20? I am already using duinoDCX on some WT32-ETH01 boards but without the ethernet activated. I can also try and help with developing the firmware but I'm not so firm yet with the arduino wifi / server frameworks to know where to start.