lasselukkari/DuinoDCX #249: Add MUTE function for SUM

soundavinci, 2020-12-12:
On the level overview and Mute tab the SUM chain is missing. Muting e.g. A+B inputs is not muting the SUM chain.

It would be nice to have also the SUM chain accessible like the inputs ABC.  

Create project!

lasselukkari, 2020-12-12:
I have not even noticed before that the device actually has a function to mute the sum channel. If it works like the rest of the channels there it should be easy to implement.

lasselukkari, 2020-12-12:
This turned out to be easier than expected. It's just missing from the ui.

lasselukkari, 2020-12-12:
[image]

The feature is now available with the latest release: https://github.com/lasselukkari/DuinoDCX/releases/tag/v0.0.39

closed 2020-12-12 by lasselukkari

soundavinci, 2020-12-12:
Hello Lasse,
That was fast. Thanks a lot. I will test the version in the next days!
Thanks again for the great effort providing this SW solution for the DCX!

BR soundavinci

PR #250 (2020-12-12): Update sum mute UI
[image]

Add missing sum channel mute button. Fixes #249.