lasselukkari/DuinoDCX #256: Dcx2496 stucks after plugging to DuinoDcx

sonikkkk, 2021-11-13:
**DCX2496 Stucks when plugged to duino dcx**
when plugged to Duino Dcx my Dcx2496 stucks with some red leds on.
Sometimes gives me error saying “Store on card error”, other times I get the error “Too much eq bands”. Dcx2496 stucks every time with both of 2 errors.

Wifi works correctly.
I can connect to Dcx2496 wifi and see the interface on “searching” mode.
Tried to use other jumper cables, tried to change Rs232/TTL module  and tried to change Rs232 null cable.
I assembled 2. Followed same wiring guide. Followed same way to upload sketch.
One works and other one doesn’t work.
I’m using firmware 0.37.

Someone experienced same issue?
Thanks in advance.

lasselukkari, 2021-11-14:
Hello! I have never heard about anything like this before. I would estimate that there is a few hundred people who have built this project. Sounds like that there is something wrong with the hardware when the other one works and the other doesn't.

You said you have switched Rs232/TTL module. Does the problem persists with the Rs232 module or with the microcontroller? Either of them are so cheap that I would just get a new one.

sonikkkk, 2021-11-14:
Thanks for your help. I switched rs232 cables, Null adapter, jumper cables but the problem persist. The other one works with parts switched. I think that Esp32 has some hardware fail. I already said it to the seller to know if other customer experienced the same issue.
I will try with another Esp32 board.
Thanks

sonikkkk, 2021-11-14:
Solved.  I got others Rs232/TTL modules from a local seller. The problem is on the Rs232 connector.

closed 2021-11-14 by sonikkkk