lasselukkari/DuinoDCX #257: WiFi Connection

pitchbent, 2022-02-28:
**Describe the bug**
Hey there, i can not get a connection to the DuinoDCX using the serial chiller board. I can't connect via laptop at all (P14s running Manjaro), with my phone (Pixel 3A) the connection instantly resets when I try to load the web page.Is there any workaround?

**To Reproduce**
Try to connect to device.

**Expected behavior**
A connection.

**Desktop (please complete the following information):**
 - OS: Manjaro
 - Browser Firefox
 - Version 96.0.3

**Smartphone (please complete the following information):**
 - Device: Pixel 3A
 - OS: Android 12
 - Browser Firefox
 - Version [e.g. 22]

**Hardware (please complete the following information):**
 - DuinoDCX firmware version: 0.0.39
 - Development board: Serial Chiller

**Development environment (please complete the following information):**
 - Arduino IDE version: 1.8.19
 - ESP32 Arduino core version: 2.0.2
 - Upload speed: 921600
 - Partition Scheme: Minimal SPIFFS (1.9M APP with OTA/190KB SPIFFS)

lasselukkari, 2022-03-01:
I have not heard about such problems before. 

It seems that you are not using the ESP32 core version 1.0.4 defined in the instructions. I have not tested the the latest 2.X.X version but previously it had problems.

I the problem persists with core version 1.0.4 you could try to some of the web server examples that come with the Arduino ESP32 core. Try this one first: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/SimpleWiFiServer/SimpleWiFiServer.ino. If it works try modifying it so that it creates a wifi access point. 

If you have an ESP32 development board I would try the examples and the DuinoDCX firmware with that too to make sure your hardware is working as expected. The connection to the ESP32 development board with DuinoDCX firmware should work even without any of the other hardware connected.

pitchbent, 2022-03-03:
You are the best! Seems like I got a little bit tired when I tried to flash the board... now it seems to work. Now I only have to connect it to my Ultradrive. Thank you for this amazing Soft- & Hardware!

closed 2022-03-03 by pitchbent

lasselukkari, 2022-03-03:
I'm happy you got it working!