lasselukkari/DuinoDCX #259: Automatic wifi connection after reboot

djGoofy, 2022-06-13:
Hello, 
is it possible that the Arduino automatically connects to a network after startup to which it was already connected. In my case is a router which starts at the same time as the Arduino. Unfortunately, I have to set up the connection every time. In addition, I have to try 3-5 times before a connection can finally be established. The error Connection Failed always comes first. otherwise everything works very well.

lasselukkari, 2022-06-14:
Hello! 

You can increase the time the device waits for the wifi connection at the startup here: https://github.com/lasselukkari/DuinoDCX/blob/master/DuinoDCX/Config.h#L25

The reason why it does not try create the connection if WiFI is not found at startup is that it made the whole thing really laggy. It will try to reconnect if the connection was there, but back when I was developing this it made the whole system unstable if it kept on trying to reconnect forever.

If the longer connection timeout does not solve the problem you  may uncomment this line of code here:
https://github.com/lasselukkari/DuinoDCX/blob/master/DuinoDCX/DuinoDCX.ino#L400

It should then try to keep on connecting forever but may have unwanted side effects.

djGoofy, 2022-06-14:
Thank you for the quick feedback. I'll be honest, I didn't expect the answer so quickly. I will test this in July. 
Greetings from Germany