lasselukkari/DuinoDCX #260: Weboberfläche hängt in der Suche fest

trischi01, 2023-08-10:
Hallo, ich habe mir das DuinoDCX geflasht. Es hat auch alles soweit funktioniert und habe auch alles richtig verbunden. Aber wenn ich die Website mit der IP-Adresse öffne steht mir da Searching und er findet die DCX nicht kannst du mir da bitte helfen?

Mit freundlichen Grüßen

clatoverada, 2023-08-11:
I had the same problem. Several RS232 adapters broke during the test phase. This probably always happened when the ESP32 was switched off. I have now simply soldered a 1k resistor into each of the 2 data lines from the RS232 Adapter to the DCX2496. Since then, the RS232 adapters have not broken any more.

lasselukkari, 2023-08-16:
For some reason the connection between the mcu and the dsp is not working. Swap parts or debug in some other way until you know what the problem is.

closed 2023-08-16 by lasselukkari