lasselukkari/DuinoDCX #264: issue with loading dcx website

Kvjavsky, 2025-02-09:
hello, I have a problem with loading the page from dcx, after completing all the activities with programming the esp and connecting it to dcx, setting the rs232 port and connecting to the esp via a computer, the page can load for up to half an hour without any results, it only displays the inscription "searching" all the time.

lasselukkari, 2025-02-10:
Please read all the closed issues about this topic.

closed 2025-02-17 by lasselukkari

Kvjavsky, 2025-02-21:
> Please read all the closed issues about this topic.

I have read all the above problems, solved and not, and none of them helped me, I installed esp32 on two different boards and the same problem still exists, and the only thing I don't understand from the manual is downloading esp32 core 1.0.4 from the link provided there, after entering the link, I see a directory with 3,000 files and I don't know which one I should download because none of them is described as in the manual, plus after entering the arduino program, the only The available ports that I can see for installation are: COM3, COM4, ​​COM5.

lasselukkari, 2025-02-21:
You need to install it in the Arduino IDE and compile the project with that.

https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

Kvjavsky, 2025-02-21:
> Należy zainstalować go w Arduino IDE i skompilować projekt za jego pomocą.
> 
> https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

ok, thanks, I already had it installed but now I understand what it's about, I changed the cor to 1.0.3 and I'm trying to do it on another board, thank you for your help