lasselukkari/DuinoDCX #265: website wont show

sethunD123, 2025-06-02:
so, i upload the code to esp32 board and i goto the https://192.168.4.1/ and it shows loading which is good sign but the user interface never loads can you help me 

I used the correct settings im sure about that 
the correct esp32 core also

HumbleDeer, 2025-06-06:
Try these steps in this order to help us troubleshoot:

1. Disconnect the ESP32 from the DCX entirely
2. Connect the ESP32 board to your computer with either the USB port or the serial port (using a USB<->UART bridge)
3. Open a serial terminal on the virtual COM port using PuTTY, your Arduino IDE, or something like SimplySerial with the -autoconnect option
4. Report back whether the serial terminal has any output, and if yes, what the output is. 

I suggest using SimplySerial (Standalone) with the -autoconnect:ANY flag for your case, which will auto connect to the serial terminal when it pops up as a device after plugging in. That'll let you catch even the very first messages, or close to. To use SimpleSerial, just open a terminal/command prompt in the folder where the binary file exists (for Standalone version).

plungoci, 2025-06-07:
I got the same issue :(

<<< SimplySerial v0.9.0 connected via COM6 >>>
Settings  : 9600 baud, no parity, 8 data bits, 1 stop bit, UTF-8 encoding, auto-connect on, echo off
Device    : VID:10C4 PID:EA60
---

Use CTRL-X to exit.

Tx)l��ͽ�I�Uh��RU���q�PU���q�PU���q��U��Tq�ReU��K�Uh��RU���q�TU���q�VU���q��U���q��eU��+�Uh�

lasselukkari, 2025-06-09:
This topic has been covered countless times in the past issues. The connection between the esp32 and the dcx2496 is not working for some reason. Please read the closed issues.

lasselukkari, 2025-06-10:
> btw how i make this ip address accessive from anywhere in the world? can you give me an idea how to do that maybe 😊

Nice to hear you got it working. I'm so happy this project still has some many users after all these years. 

You could set up a port forwarding rule to your router and expose the device to the public internet. A VPN would be a lot safer solution.

plungoci, 2025-06-10:
Solved my issue, it seems that both rs232 to ttl with max3232 that i bought won't change the RX and TX by themselves :/ so in order to make it work i made a null modem and swap RX and TX now it is working like a charm <3

Maybe @lasselukkari you can add this tip on the instruction to add a null modem

[image]

lasselukkari, 2025-06-10:
> Maybe [@lasselukkari](https://github.com/lasselukkari) you can add this tip on the instruction to add a null modem

Did you read this: https://lasselukkari.github.io/DuinoDCX/

It says:

 _Male to male null modem cables are really hard to find. Because of this we need to use a male to male gender changer combined with a female to male null modem cable. Do not buy a cable if you are not sure that it is a null modem cable._

sethunD123, 2025-06-10:
Hey man do you have upcoming new updates to this project because lot of people in this field still use this thing.
My dcx is also 15+ years older work like charm 😊

I really apperciate of you can give some updates to this software

HumbleDeer, 2025-06-29:
None of the USB to UART converters swap RX and TX for you, because that's not part of the specification and they should in no case be doing something like that. It would be a monumentally terrible idea to introduce such undefined nonstandard behaviours.

sethunD123, 2025-07-02:
> None of the USB to UART converters swap RX and TX for you, because that's not part of the specification and they should in no case be doing something like that. It would be a monumentally terrible idea to introduce such undefined nonstandard behaviours.

Yes you are right

closed 2025-07-02 by sethunD123

lasselukkari, 2025-07-02:
> [@lasselukkari](https://github.com/lasselukkari) do you mind to expand the software?

I'm sorry but I don't understand what you mean?

sethunD123, 2025-07-05:
> > [@lasselukkari](https://github.com/lasselukkari) do you mind to expand the software?
> 
> I'm sorry but I don't understand what you mean?

do you planing to add some new features?

lasselukkari, 2025-07-06:
> do you planing to add some new features?

Not really. Of course, there are always things that could be improved, but I consider the project feature-complete at this point. It would be nice to have the ability to save and load presets, but figuring out how to implement that would take quite a bit of time. Lately, I’ve been enjoying spending time outdoors and working on my pond more than sitting in front of a computer for long hours. That said, contributions to the project are more than welcome!