lasselukkari/DuinoDCX #267: i can't connect to the esp network

Krystian252, 2025-07-10:
I have Arduino ESP 1.0.4 and I can't connect to the ESP network. If I update to the latest version, I can, but then I get "searching" all the time. Can you help me?

lasselukkari, 2025-07-14:
> Can you help me?

Without further information no. Did you read this: https://github.com/lasselukkari/DuinoDCX/issues/266.

Krystian252, 2025-07-14:
Yes, but I can't connect to the ESP. The ESP network is set to "DCX2496," but after entering the password, it tries to connect but gets an error message. After installing ESP 3.2.1, I was able to connect, but it kept saying "searching."

Krystian252, 2025-07-14:
[image]
[image]

[image]
[image]
[image]

Krystian252, 2025-07-19:
I really need a remote control for my DCX

clatoverada, 2025-07-20:
From what I can observe, I possess an identical RS232 adapter to the one you're utilizing. I constructed two adapters using this one, yet each functioned only a single time. During the initial test, I successfully established a connection right away. Following that initial test, I could no longer form a new connection. I believe the RS232 adapter was damaged when disconnected from the DCX while under voltage. To prevent damaging the adapter during disconnection, I soldered two resistors into the serial cable and replaced the RS232 adapter with new units. Since then, everything has been operating without issues.

clatoverada, 2025-07-20:
[image]

clatoverada, 2025-07-20:
The bridge (blue connection on the left)  between pin 7 and pin 8 on the RS232 adapter side is also important. Without this bridge it does not work for me:

[image]

clatoverada, 2025-07-20:
So I now understand that you cannot establish a connection to the access point 
send out by the ESP32 with the SSID “DCX2496” and Password "Ultradrive".

Correct?

I do this myself with an old smartphone, here is a screenshot of the connection:

[image]

Krystian252, 2025-07-20:
No, i can't connect to esp WiFi

clatoverada, 2025-07-20:
Can you explain this more precisely for us outsiders? A simple "No, i can't connect to esp WiFi" is not enough to help you.

Krystian252, 2025-07-20:
I'm trying connect to the "DCX2496" WiFi, but I can't 

[image]

Krystian252, 2025-07-20:
iPad, I tested on iPad, iPhone, pc, Mac, Android tablet and there is the same problem :/

clatoverada, 2025-07-20:
That's really unusual. I've never seen this before in connection with the ESP32 on my devices.

If I enter "unable to join the network" in Google's search for pictures, there are many such reports to be found. Maybe this will help you?

Krystian252, 2025-07-20:
I still can't connect to the ESP :/

Krystian252, 2025-07-30:
@clatoverada can you make me one ready duinoDCX, and send to me? I really need it for my job

clatoverada, 2025-07-30:
I don't think a new device will solve your problem with the Wifi.