lasselukkari/DuinoDCX #268: Fast help!!!

Igornik955, 2025-07-24:
@lasselukkari . I have a big problem with duinodcx i tried everything from github even from other forum and nothing helped. Can you send me a complete ESP32 board ready to start duinodcx. I need fast help because i need this to work immediately. Pls help.

lasselukkari, 2025-07-25:
Unfortunately I don't have any for sale.

Igornik955, 2025-07-25:
> Unfortunately I don't have any for sale.

When you will optionally get it and sell it to me because i'm really need it.

clatoverada, 2025-07-26:
As I said, I don't think it's a problem with the ESP32. Unfortunately, I haven't seen this error message myself yet. But I think if the “DCX2496” network is selectable but the connection doesn't work, then the problem may be due to a setting in the smartphone. Maybe you can try it with other smartphones, preferably one belonging from a other Person?

Igornik955, 2025-07-26:
> As I said, I don't think it's a problem with the ESP32. Unfortunately, I haven't seen this error message myself yet. But I think if the “DCX2496” network is selectable but the connection doesn't [...]

I already tried with another device with ios and android and same problem. The connection only working with latest ESP32 core version but on latest version there is problem with seraching. ESP dont see the dcx. This problem is very uniqe and i dont think what to do next.

clatoverada, 2025-07-26:
I have integrated the two resistors directly into the sub-d enclosure.

Igornik955, 2025-07-26:
> I have integrated the two resistors directly into the sub-d enclosure.

Ok, i will do a bridge again and i will ad a 2x 1K resistors and i will run it on a latest ESP32 core version